open-rdc/orne-box #3: NVIDIA Jetson Xavier のセットアップ

yasuohayashibara, 2020-03-10:
NVIDIA Jetson Xavier のセットアップの方法を確認してメモを残す．

yasuohayashibara, 2020-03-10:
出村先生が執筆したインストールの方法を発見．
参考にする．
https://demura.net/athome/15393.html

yasuohayashibara, 2020-03-10:
@taishiyamamoto @tiger0421 
セットアップに関して知見が得られたらこちらに追加してもらえればと思います．

taishiyamamoto, 2020-03-11:
出村先生が執筆したセットアップの方法はJetPackのバージョンが4.1.1ですが，自分たちは4.3でセットアップを行いました．
基本的な捜査の流れは同じだと思いますが，GUIに大きく変更が加わっています．
https://docs.nvidia.com/sdk-manager/install-with-sdkm-jetson/index.html

taishiyamamoto, 2020-03-11:
セットアップの方法を記入しました．
https://docs.google.com/document/d/1yr4Er2KtaCM1qDe_Tm_X0hn2kUaMX4Est1JboUNOacU/edit?usp=sharing

**インストールできたホストPC環境**

- Lenovo ThinkBook13s(Intel® Core™ i7-8565U, Memory 16GB, GPU Intel® UHD Graphics)
- Ubuntu 18.04.4 LTS
- kernel 5.3.0-40-generic
- gcc/g++ 7.4.0

- Lenovo ThinkPad X240(Intel® Core™ i5-4300U, Memory 8GB, GPU Intel® Haswell Mobile)
- Ubuntu 18.04.4 LTS
- kernel 4.15.0-76-generic
- gcc/g++ 7.4.0

yasuohayashibara, 2020-03-12:
Jetson Xavier agx のインストール
所要時間:約2時間
1.	nvidia_sdk_managerのサイトへ行く
https://developer.nvidia.com/nvidia-sdk-manager

[image]

2.	「Nvidia SDK Manager」をクリックすると，ログインすることを要求される．

[image]

3.	「log in」をクリックするとアドレスとパスワードの入力を要求されるので自分のアカウントでサインインする．（「join now」を押すと会員登録ページに飛ばされる）

[image]

4.	サインインするとマイページに飛ばされるので，上部にある「Nvidia SDK manager」をクリックする．

[image]

5.	押すと.debファイルをダウンロードできるので，ソフトウェアのインストールを行う． 

[image]

6.	Nvidia SDK managerを起動する．

[image]

7.	起動するとサインインを要求されるのでサインインする． 

[image]

8.	　STEP 01で対象のハードウェア(Jetson Nano Jetson TX2 Jetson AGX Xavierなど)を選択できる．今回はJetson AGX Xavierを選択し，「CONTINUE」をクリック

[image]

9.	STEP 02でインストールするパッケージを選択できる．ページの下部に利用規約に同意するためのチェックボックスがあるのでそれをクリックし，「CONTINUE」を押す．  押すと19GB程のダウンロードが始まるので，ホストマシンの空き容量を確認する．

[image]

10.	 ダウンロードが終わりインストールが始まり，しばらく経過すると，Jetsonのインストール方法を決める画面がポップアップする．

[image]

11.	ポップアップした画面にも書いてあるが，セットアップの方法を「Automatic Setup」から，「Manual Setup」にする．

[image]

12.	ここからはJetson AGX Xavierが必要になるため用意する．（ダウンロードを行っている間に行う方が時間の効率は良い）

13.	Jetson AGX Xavierを操作できる状態にする．｛セットアップにはACアダプタ（付属），キーボード，マウス（USBのポートが少ないのでハブが必要），インターネット（有線接続），モニター（HDMI），ホストマシンとXavierを繋ぐケーブル(付属の USB-TypeC to A)が必要｝

14.	接続後，Jetson AGX Xavierをセットアップできる状態にする．
まず，Force Recovery Buttonを押した状態を維持し，Power Buttonを押す．そして，Force Recovery ButtonとPower Buttonを離すという手順がいる．

[image]

15.	成功すると白い電源ランプが光り，ホストマシンでlsusbコマンドを叩くとNvidiaのデバイスが確認できる．
確認できたら，「Flash」をクリックする．

[image]

[image]

16.	しばらくすると，Jetson AGX Xavierが起動し，Ubuntuの初期設定の画面が出てくる．設定を完了すると，サインインできるようになる． 

[image]

17.	その後，Jetson AGX XavierのIP AdressとUsername，Passwordの入力を要求されるので，先程設定したものを入力し，インストールする．
(IP Adressはデフォルトの192.168.55.1で問題なかった．)  

[image]

18.	インストールが問題なく完了すると，以下のような画面になる．「FINISH」をクリックして終了する．

[image]

参考
●	Jetson AGX Xavier Developer Kit User Guide　https://developer.nvidia.com/embedded/dlc/jetson_agx_xavier_developer_kit_user_guide

yasuohayashibara, 2020-03-12:
@taishiyamamoto ROSはインストールしました？

tiger0421, 2020-03-12:
installROS.shをたたくだけでインストールできました．
https://github.com/jetsonhacks/installROSXavier

yasuohayashibara, 2020-03-13:
問題あれば再度開くとしてissueを閉じます．
ご協力いただいたみなさまありがとうございます．

closed 2020-03-13 by yasuohayashibara