open-rdc/orne-box #4: 電子回路の設計

yasuohayashibara, 2020-03-11:
電子回路を設計する．

### 要求仕様
- 最低3時間の駆動時間
- PCをシャットダウンすることなく充電
- 非常停止してもロータリエンコーダを計測
それほど大きくない力で押すことができる．
- 汎用電源としてAC100Vを提供
- 汎用の部品でできるだけ構築

yasuohayashibara, 2020-03-11:
使用する部品

品名 | メーカ | 型番 | 特性
-- | -- | -- | --
バッテリ | Long | WP12-12 | 24V(12V x 2), 12Ah =  288Wh
モータ | ツジ電子 | TF-M30-24-3500-G30R
モータ | ツジ電子 | TF-M30-24-3500-G30L
モータドライバ | ツジ電子 | TF-2MD3-R6
コンピュータ | NVIDIA | Jetson AGX Xavier | max. 30W
USBハブ | iBUFFALO | BSH4U21U3BK
無線LANルータ | BUFFALO | WSR-1166DHPL/N 11ac
ブレーカ | アムトランス | CB-IBP-1-10A
非常停止スイッチ | OMRON | MNC-103
メインスイッチ | NKK | S-331T
スピーカ | HERDIO | 8541747086
タブレット | SONY | Xperia Z4 Tablet
測域センサ | HOKUYO | UTM-30LX-EW
DC/ACインバーター | セルスター | HG-350/24V
USB-C オーディオ | CableCreation | Type C外付け サウンドカード USB C 3.5mm ミニ ジャック
オーディオアンプ | ELEGIANT | ELEGIANTejxsZggFs

-

yasuohayashibara, 2020-03-11:
制御回路の構成（案）
- 各装置の電源はAC100Vで供給
- 電源スイッチは情報系と駆動系で分ける．
- 非常停止を押してもエンコーダは計測できる．
このモードで地図を生成する．
[image]

yasuohayashibara, 2020-03-11:
### モータ遮断回路

非常停止時に軽くブレーキがかかる仕様になっている．
カートとして動かす分にはブレーキがじゃまになるかも知れない．

https://github.com/open-rdc/orne_navigation/files/3929867/orne_a_PowerToMotor.pdf

[image]

yasuohayashibara, 2020-03-11:
ここに非常停止の議論があった．
これを参考に設計されたものと思われる．

http://t-frog.com/forums/forum.php?ml=icart-mini-devel&read=16

yasuohayashibara, 2020-03-11:
台車を押して地図生成するときに，以下を使用できないか検討する．
その前にもう少しドライバ周辺を私が理解しないと進められなさそう．

Spur_free()

yasuohayashibara, 2020-03-11:
[open-rdc/icart](https://github.com/open-rdc/icart) のドライバでもSpur_free()が使用されている．
台車を押す場合のために，Spur_free()を使用するdriverを用意した方が良いかも知れない．

https://github.com/open-rdc/icart/blob/ad2c49478c06de4e2636f23e74b94f4165a378fa/icart_mini_driver/src/icart_mini_driver_node.cpp#L82

電源を入れていないときの移動を考えると，やはりモータ遮断回路は必要そうです．

yasuohayashibara, 2020-03-11:
利用しないつもりでしたが，モータドライバ用の電源のためにも以下の電源は必要そうです．
TF-PW36-5/12M
http://t-frog.com/products/power_supply/
URGの電源はこちらから供給します．

yasuohayashibara, 2020-03-12:
以下のインターフェイスを後ろの板に取り付ける．

品名 | メーカ | 型番
-- | -- | --
コネクタ | ミスミ | MMS3102A-16-10-P
コネクタキャップ | ミスミ | MMS25042-16D
LED | 坂詰製作所 | DW-10ML-IP-AC/DC5V-60V-R/C
スイッチ | NKK | S-1AW/B
充電ケーブル | ミスミ | M3106B-A1-B-RF-2
電圧計 | KIMISS | KIMISS 2.5-30V

yasuohayashibara, 2020-03-12:
外部とのインターフェイスの配置（案）
意見があればお願いします．

[image]

yasuohayashibara, 2020-03-12:
以下にあるモータ遮断回路を製作する．
https://github.com/open-rdc/orne_navigation/wiki

部品としては，以下となるがどうして12V?
あと，サージ対策用としてダイオードを入れたほうが良いかも知れない．

- コイル　LY2 DC12
- セメント抵抗　20W 10Ω

shulace13, 2020-03-12:
そのモーター遮断回路は
https://github.com/open-rdc/orne_navigation/issues/341
に書かれているようにモータドライバに入る電源にキャパシタを挟むことで解決できます。
12V入っているのはリレー駆動用と思われます。あるいはPowerSupplyBoardから引っ張ってきやすかったからなのかなとも思います。

yasuohayashibara, 2020-03-12:
議論に加わってくれてありがとう．
ここは色々と複雑な事情があります．

### モータドライバの故障の回避
モータドライバの電源に**低ESRキャパシタ**を入れるのは，モータドライバを壊さないためです．
http://t-frog.com/products/motor_driver/files/TF-2MD3-R6_Manual_rev5.pdf
急激に電源を遮断したときでも，モータドライバが壊れなくなるようです．

### モータドライバの電源の遮断による非常停止
上記の対策により，モータドライバの電源を遮断することで非常停止できます．
ただし，押して動かすことには向きません．
電源遮断時に台車を動かすとモータが発電して，モータドライバに電流が流れ込みます．
この電流が大きいと，モータドライバを故障させるおそれがあります．
また，台車を押すときの力も大きくなります．
（人追従のロボットはこちらを採用しているので動かすのが大変です．）

### モータドライバのフリーモード
モータドライバにはフリーモードがあります．
フリーモードを選択すると，発電した電流はモータドライバに流れません．
台車を押す力は小さくなり，モータドライバの故障の懸念も無くなります．
ただし，5Vが供給されていて，USBからコマンドを受信することが必要となります．
つまり，電源を入れてコマンド送らないと，このモードにはなりません．
ロボットを動かす前に，ちょっとだけ動かしたい時には利用できません．

### モータとモータドライバの断線
電源を入れずに押して動かすためには，モータ遮断回路が必要です．
なお，単に遮断すると減速すること無く動き続けるため，非常停止できません．
そのため，10Ωのセメント抵抗を使用してブレーキがかかるようにしています．

### モータ遮断回路のノイズ
モータを遮断するために，回路にはリレーが使用されています．
リレーはインダクタンス成分があるため遮断時にサージを発生します．
サージ除去のために，ダイオードなどを入れないと回路に高電圧が加わります．
これにより周辺回路が故障するおそれがります．

### 今後の方針
- 非常停止
従来どおりモータ遮断回路を使用する．

- 地図生成などで長時間押して動かす時
非常停止の状態で動かすと，モータや10Ωの抵抗で無駄なエネルギを消費します．
長時間押して動かすときには，モータドライバをフリーモードにしたほうが良いです．

以上です．

と書いていて気づきましたが，

- 非常停止
モータドライバの電源の遮断
- モータ遮断回路

と２系統にすると良いかも知れません．

- 通常の非常停止はモータドライバの駆動用の電源を遮断してブレーキをかける
- 長時間押すときには，モータとモータドライバの間を遮断する
電源を入れていないときには，そもそも遮断した状態にする．

と良さそうな気がします．検討します．

yasuohayashibara, 2020-03-13:
**非常停止**と**押して動かす**ことを両立できる回路です．
ただし，10A以上流れたときにPCを含む電源が遮断されるので，もう少し工夫をしたほうが良いかも知れません．

[image]

使用するリレーとソケット

品名 | メーカ | 型番
-- | -- | --
リレー | OMRON | LY3N-D2 DC24
リレーソケット | OMRON | PT11 FOR LY
リレー | OMRON | LY1N-D2 DC24
リレーソケット | OMRON | PT08 FOR LY

yasuohayashibara, 2020-03-13:
以下のような無停電電源を使用すると良いかも知れません．
1.9kgと重いですが．．．

WOW-300R

yasuohayashibara, 2020-03-13:
設計が終了したので，issueを閉じます．
最終的な回路の構成に関しては，動作確認しながら決定します．

closed 2020-03-13 by yasuohayashibara

yasuohayashibara, 2020-03-13:
追加です．

本日3Fのロボットを押して動かしてみましたが，モータとモータドライバを接続しても，それほど抵抗なく動くことを確認しました．
電流が電源側に逆流して電源のランプが付きますが，ブレーキモードになっていなければ，しばらく動かすのは問題ないのかも知れません．

安全を考えると遮断回路はあったほうが良いですが，無くてもそれほど問題が無いかもしれません．