open-rdc/orne-box #8: JetsonのGPUの評価

tiger0421, 2020-03-22:
GPUの評価をするためにKerasのYOLO-v3を動かす．

- [ ] Tensorflow-gpuのインストール
- [ ] GPUを認識するか確認
- [ ] YOLOを動かす

## 参考
https://qiita.com/yoyoyo_/items/10d550b03b4b9c175d9c

tiger0421, 2020-03-22:
## 結論
なんとかインストールできた．

## 失敗
https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html
上のURLの通りにやってもうまくいかなかった．

## 原因
* ScipyやCythonのビルドに失敗するのでpip3でインストールした．
* 他にもいろいろインストールしてしまったので本当にpip install だけが原因かはわからない．

tiger0421, 2020-03-22:
Tensorflow2.1.0からGPUを認識するか確かめた．
`from tensorflow.python.client import device_lib`
`device_lib.list_local_devices()`

画像下の方にGPUの項目があるので，GPUの認識はできていることが分かる．
[image]

tiger0421, 2020-03-31:
容量が全部で30GBもない．
現在，残り約１GB．
ここまでくると，Tabキーでの補完など，一時ファイルの生成に問題が出るため操作に支障をきたす．
~~なので好きなものをポンポン入れていくとあっという間に容量がなくなる．~~
→HDD増設することで対処

インストールしたもの
ホスト
- ROS melodic
- Tensorflow-gpu (python3)
- Docker (GPUが認識することは確認済み)
    - Jetson用のコンテナ（ROS melodic & Tensorflow-gpu & tf-pose-estimation インストール済み）

今後，YOLOやtf-pose-estimationを試し，python2でインストール&動作確認を行う予定．

NaokiSato102, 2020-03-31:
>HDD増設

筐体さえ用意してしまえば解体したPCの物のが使用可能かと思います。
筐体無し（もしくは適当に自作）でも可能でしょうけどもちょっと心配です。

tiger0421, 2020-03-31:
> ちょっと心配です。
心配事はわからないけど，増設の可否についてなら問題ないかと．
M.2を増設している人はいます．
https://medium.com/@ramin.nabati/installing-an-nvme-ssd-drive-on-nvidia-jetson-xavier-37183c948978

yasuohayashibara, 2020-03-31:
M.2 良いですね．
256GBで良いですかね？

tiger0421, 2020-03-31:
地図を作成するなどの際は，データをパソコンの方に保存するということでしたらぎりぎりいけるんでしょうか...?
ちなみに，自分のデスクトップは30GB，ラップトップでは60GBのスペースを消費しています．
研究室の学習サーバーには，/varや/tmpも含めて463GB使用されています．
そのうち/homeが232GBなので約230GBがシステムやアプリケーションに使われています．

この辺りは地図のデータをとるかなどを考えたうえで決めたほうがいいと思います．

NaokiSato102, 2020-03-31:
解体済みPCから流用するHDDの件での心配事は、規格の一致と防振防塵性の観点で心配しています。
M2SSD搭載を現在検討中とのことですが、現状をざっと見るに、ちょっと256GBでは不足ではという話になって良そうなので、USB接続外部HDDとしての搭載の方面も検討してみてはいかがかなと思いました。

yasuohayashibara, 2020-04-05:
以下に基づいてyolov3-tinyの調査を行った。
https://www.nakasha.co.jp/future/ai/vol9_yolov3nvidia_jetson_agx_xavier.html

以下、判明したこと
- 画像を表示しなければ100FPS程度で処理が可能
- 100FPSのCPUの負荷は170%程度
- 画像を表示しながらだと40FPS程度
- yolov3-tinyは誤認識はあるが、人の検出は優秀

以前つくばチャレンジで撮影した画像でチェック
```
./darknet detector demo cfg/coco.data cfg/yolov3-tiny.cfg yolov3-tiny.weights 2018\ ORNE.mp4
```

ビデオ
https://youtu.be/-KmUT4xNtoI

yasuohayashibara, 2020-04-05:
調査したためissueを閉じます。

closed 2020-04-05 by yasuohayashibara