open-rdc/orne-box #26: 防水加工

IGC8810, 2021-07-23:
雨天でも走行できるようにする。

maHidaka, 2021-07-28:
7/28日に防水処理の作業、実験を行った

- 以前の防水性能評価の実験でアルミフレームの隙間から水が侵入する問題があった
   - 防水の確認 #20

maHidaka, 2021-07-28:
# 作業内容

- アルミフレームの隙間埋め
   - 材料
      - [シールカバー(HSCK5-B-12)](https://jp.misumi-ec.com/vona2/detail/110302695550/?ProductCode=HSCK5-B-12)
      - [溝カバー(HSCJ5-B-12)](https://jp.misumi-ec.com/vona2/detail/110302493950/?ProductCode=HSCJ5-B-12)
   
  - 外部から内部につながる溝をシール材で埋めた
    - [image]
    - 底板、側面上左右の溝に同様のシール材を入れた

maHidaka, 2021-07-28:
- 天板部分の防水処理
　- 前回の防水確認では天板からの浸水が確認されたため天板上部に溝カバーを一周させて天板を密着させる形で取り付けた

[image]

maHidaka, 2021-07-28:
- ハンドル部分の溝カバー取り付け 
  - 内側の面のみシール材
[image]

maHidaka, 2021-07-29:
# 防水テスト結果
- 実施場所
   - 津田沼1号館前の水場
   - [実験のようす(動画](https://youtu.be/GXSvCJiYqCM)

- 結果
   - アルミフレームの溝にシール材を詰めた部分について、アルミフレームの溝から水が流れ込んでくることはなくなった
   - 天板については浸水は見られなくなった
   - **アルミフレームの接合面から水が侵入してくる水は防げない** 
   - アルミフレームの一部から浸水すると毛細管現象によってフレーム全体へ広がっていく
  
- 顕著な浸水箇所について
   - ハンドル底部のアルミフレーム接合部からの水の侵入が見られた
   [image]
   - 同様にフロントパネルのアルミフレーム接合部からも水の侵入があった

maHidaka, 2021-07-29:
- アルミフレーム同士の接合部に薄いゴムシート等を挟んで組み付ける
- シリコンコーキング等で隙間を塞ぐ(分解組み立て毎に作業する必要があり)

等の対策を検討

底板、サイドパネルなどの板は長辺3点で締結されているが、十分な密閉を得ようとして今回のような形でアルミフレームと板の間にシーリング材を挟むと板がシール材の反発力で膨らんで変形することがわかった
板厚を上げるか、締結箇所を増やす等の対策も検討する
フレームと板の間は[こんな感じのパッキン](https://jp.misumi-ec.com/vona2/detail/110302683740/?KWSearch=%e3%82%b4%e3%83%a0%e3%83%91%e3%83%83%e3%82%ad%e3%83%b3&searchFlow=results2products&relatedKeyword=1)でシールするのが理想

maHidaka, 2021-08-10:
## 7/30
### 追加工

- 防水の追加工として浸水が見られた隙間にシリコンコーキング剤を使用
   - フロントパネル隙間
   -  バックパネル隙間
   -   底板隙間
 
- コーキングの外観
[image]
- 
[image]
[image]

## 防水試験

- ホースから機体全体、前回浸水がみられた部分に対して直接水をかけた
- ５分ほど水をかけて内部への浸水状況を確認する

- 防水後の様子
  - 前回の浸水部も含めて、一切の浸水は見られなかった

maHidaka, 2021-08-10:
定性的な実験となるが、現実的な防水性能が得られたことが確認できたのでクローズします

closed 2021-08-10 by maHidaka