open-rdc/orne-box #29: GNSSレシーバの実装

IGC8810, 2021-07-25:
ナビゲーションに組込むまでの過程をWikiにまとめる

taishiyamamoto, 2021-08-06:
[このリポジトリ](https://github.com/NaokiTakahashi12/connect-serialport-for-bluetooth.git)でDG-PRO1RWとの接続を試みたができなかった。

* 実行したコマンド
```
sudo sh connect.sh
```
* エラー
 ```
Can't get device list: Operation not supported
Can't create device: Operation not supported
```

taishiyamamoto, 2021-08-06:
現在,GNSSレシーバ(DG-PRO1RW)は[RFCOMM](https://www.wdic.org/w/WDIC/RFCOMM)という通信プロトコルを用いて通信している。
そして、JetsonのデフォルトのカーネルではRFCOMM通信に対応していないらしい。([参考元](https://forums.developer.nvidia.com/t/jetson-nano-bluetooth-issue-rfcomm-tty-support-not-available/81432))
そのため、コンフィグを設定してカーネルをリコンパイルする必要がある。

jetsonのリコンパイルの方法は下記サイトが参考になりそう。
https://blog.hypriot.com/post/nvidia-jetson-nano-build-kernel-docker-optimized/

taishiyamamoto, 2021-08-24:
下記サイトを参考に作業を行なったがカーネルのビルドが通らなかったので，SCR-u2t(NEO-M8P搭載)の使用を検討する．

taishiyamamoto, 2021-08-24:
下記パッケージを用いて動作確認を行う．
https://github.com/KumarRobotics/ublox

taishiyamamoto, 2021-09-01:
- command

```
roslaunch ublox_gps ublox_device.launch param_file_name:=test
```
- parameter file (test.yaml)
test.yaml
```
# Configuration Settings for C94-M8P device

debug: 4                    # Range 0-4 (0 means no debug statements will print)

save:
  mask: 3103                # Save I/O, Message, INF Message, Nav, Receiver 
                            # Manager, Antenna, and Logging Configuration
  device: 4                 # Save to EEPROM

device: /dev/sensors/gnss
frame_id: gps
rate: 4                     # in Hz
nav_rate: 4                 # [# of measurement cycles], recommended 1 Hz, may 
                            # be either 5 Hz (Dual constellation) or 
                            # 8 Hz (GPS only)
dynamic_model: airborne2    # Airborne < 2G, 2D fix not supported (3D only), 
                            # Max Alt: 50km
                            # Max Horizontal Velocity: 250 m/s, 
                            # Max Vertical Velocity: 100 m/s
fix_mode: auto
enable_ppp: false           # Not supported by C94-M8P
dr_limit: 0

uart1:
  baudrate: 115200           # C94-M8P specific
  in: 32                    # RTCM 3
  out: 0                    # No UART out for rover

gnss:
  glonass: true             # Supported by C94-M8P
  beidou: true             # Supported by C94-M8P
  qzss: true               # Supported by C94-M8P

dgnss_mode: 3               # Fixed mode
tmode3: 0

inf:
  all: true                   # Whether to display all INF messages in console

# Enable u-blox message publishers
publish:
  all: true
  aid:
    hui: false
 nav:
    posecef: false
```

taishiyamamoto, 2021-09-01:
wikiにSCR-u2tのROSでの使用方法のドキュメントを書く．

taishiyamamoto, 2021-09-01:
ドキュメントを記載した．
https://github.com/open-rdc/orne-box/wiki/Set-up-GNSS-Receiver%28SCR-u2t%2C-Independent-positioning%29

taishiyamamoto, 2021-09-01:
GNSSレシーバーを実装したので共有したら閉じる．
別のレシーバーを用いる場合は再度issueを立てる．

closed 2021-10-02 by IGC8810