open-rdc/orne-box #42: オドメトリの調整

yasuohayashibara, 2021-09-25:
オドメトリの調整をするパラメータを明らかにする．

yasuohayashibara, 2021-09-25:
以下がオドメトリに関連する部分だと思います．

https://github.com/open-rdc/orne-box/blob/0e9e224db304d01cfd6f4884e86229e9804035dd/orne_box_setup/config/orne_box.param#L9-L10

https://github.com/open-rdc/orne-box/blob/0e9e224db304d01cfd6f4884e86229e9804035dd/orne_box_description/urdf/orne_box.urdf.xacro#L104

https://github.com/open-rdc/orne-box/blob/0e9e224db304d01cfd6f4884e86229e9804035dd/orne_box_description/urdf/orne_box.urdf.xacro#L142

ryuseiotsuka, 2022-10-21:
box2のオドメトリの調整
・実験内容
18階の廊下5ｍを複数回走らせホイールサイズ変更によってodomと実際に進んだ距離の差を減らす
・結果
　調整前　実環境を5ｍ走行した際のodomから計算した走行距離　平均5.26m(7回実験)
　誤差5.2%
　調整後　実環境を5ｍ走行した際のodomから計算した走行距離　平均5.02m(2回実験)
　誤差0.4%

[orne-box/orne_box_setup/config/orne_box.param](https://github.com/open-rdc/orne-box/blob/0e9e224db304d01cfd6f4884e86229e9804035dd/orne_box_setup/config/orne_box.param#L9-L10)
変更前のパラメータ
```
RADIUS[0] 0.152
RADIUS[1] 0.152
```
変更予定のパラメータ
```
RADIUS[0] 0.149
RADIUS[1] 0.149
```

closed 2022-10-21 by ryuseiotsuka