open-rdc/orne-box #50: コストマップ用の地図に関する検討

yasuohayashibara, 2021-11-17:
コストマップ用の地図に関して検討する．

yasuohayashibara, 2021-11-17:
コストマップを生成すると思われる地図をチェックしましたが，経路に障害物が残っているように見えます．
例えば，赤の矢印で示したところは，おそらく経路だと思いますが，障害物が残っています．
地図を取り違えているおそれもありますが，ご確認頂ければと思います．
その他，いろいろと指摘する点がありますが，まずはこちらを確認して頂ければと思います．

[image]

こちらの地図を見ました．
https://drive.google.com/file/d/1fliXT08Zal0jX6fnx_DlCgWX3qcnIfrR/view?usp=sharing

ryuseiotsuka, 2021-11-18:
オリジナルのコストマップでは経路にコストが多数残っており走行できそうになかったため修正を行いました。
https://drive.google.com/file/d/1Y2Iv-_mOcBnFrQcb2DnqOFf0WORflhGd/view?usp=drivesdk

yasuohayashibara, 2021-11-18:
おそらくこちらの部分は消し忘れているような気がします．

[image]

以下のページにrviz上でウェイポイントを編集という機能がありそうですので，ウェイポイントを見ながら経路となりそうな部分の修正を加えたほうが良いと思います．

https://github.com/open-rdc/orne_navigation/wiki/ORNE%E7%92%B0%E5%A2%83%E5%8B%95%E4%BD%9C%E3%81%AE%E6%89%8B%E9%A0%86#%E5%AE%9F%E6%A9%9F%E5%8B%95%E4%BD%9C%E6%89%8B%E9%A0%86

次の指摘ですが，走行する経路で障害物となるものを除くというのに加えて，走行してほしくないところを塞ぐということが必要です．
例えば以下は確認走行直後の道ですが，歩道と車道の境目に障害物を置いていないため，前がロボットなどで塞がれていると，車道に出ようとするおそれがあります．
大変ですが，走行してほしくない場所を塞ぐように障害物（見えない壁）を置くことが必要です．

[image]

yasuohayashibara, 2021-11-18:
従来のナビゲーションを工夫したり，低い障害物を検出できるようになっているとしたらその旨お伝えください．
その場合はコストマップは全て取り払っても構わないかもしれません．

yasuohayashibara, 2021-11-18:
以前slackで流した情報を念のためこちらにも載せます．
-----

ポイントクラウドから２次元地図への変換方法
パラメータを変更することでより良い地図が作成できると思います．
https://youtu.be/yxSkvZRwuD4

あと，低い障害物はマップに無いので，google mapの画像などと重ね合わせながら，追加することが必要です．
ついでに走行経路にあるコストは削除（消しゴムで消して）してください．
この作業は根気とセンスがいる作業で，低い障害物を検知する手段が他にない場合は，これの出来（＋ウェイポイントの位置）でほぼゴールできるかどうかが決まります．
https://photos.app.goo.gl/jfs5Qpzaj2eGiyx46 

作業途中のファイル
あとはお任せします．
https://drive.google.com/drive/folders/1hP-uqY-V0xETGWBPyDRyWDT7JXfF3FS4?usp=sharing

なお，検証していないので致命的な問題が残っているおそれがあります．
念のため学内の地図で一度試してから作業したほうが良いと思います．

サンプルでは細い線を使用していますが，もう少し太い線を使用したほうが安全かもしれません．
経路生成では，最短経路を計算して動くため，できるだけ障害物の近くを走行しようとするかもしれません．
少し自己位置がずれると脱輪します．
ただし，あまり狭めすぎると経路生成できなくなるおそれがあるため，その点は試行錯誤が必要です．

yasuohayashibara, 2023-08-23:
こちらすでに役割を終えたため，issueを閉じます．

closed 2023-08-23 by yasuohayashibara