open-rdc/orne-box #69: [GNSS] GNSSを使用した自律移動の問題

yasuohayashibara, 2023-01-30:
GNSSを使用した自律移動の問題点を議論するためのissue
大塚君関連のissueでORNE-boxには直接依存しないため，２月末には閉じる．

yasuohayashibara, 2023-01-30:
ノードの構成

[image]

yasuohayashibara, 2023-01-30:
緯度
[image]

経度
[image]

yasuohayashibara, 2023-01-30:
/ublox/fix_velocity

[image]

yasuohayashibara, 2023-01-30:
odom
位置
[image]

速度
[image]

yasuohayashibara, 2023-01-30:
/odometry/gps

速度，角速度は0rad/s
[image]

yasuohayashibara, 2023-01-30:
/odometry/filtered
tf: /odom -> /base_link

[image]

yasuohayashibara, 2023-01-30:
TFを表示してみた

[image]

yasuohayashibara, 2023-01-30:
問題点と思われるもの

１）odom -> base_link で矛盾した値が出力されている．
２）ロボットのヨー方向の姿勢が正確に推定されていない．

yasuohayashibara, 2023-01-30:
[image]

元データ
[odom.xlsx](https://github.com/open-rdc/orne-box/files/10535587/odom.xlsx)

ryuseiotsuka, 2023-02-03:
現在の状態です．
gnssの緯度経度情報のみを使って計算したodometry/gps（緑）の原点と回転方向の調整が終了しました．
odometry/gpsとIMU，ホイールオドムをekfで融合したodometry/filtered（黄）は，ロボットの実際の向きに関わらず右を向き続けてしまっています．
このロボットの方向の問題を修正すれば自律走行で使用できると思います．
[image]

ryuseiotsuka, 2023-02-03:
IMUを抜いてrobot_localizationでセンサ値を融合したところロボットの姿勢の問題が見られなかったため，IMUの各軸の向きについて調整を行う．
[image]

yasuohayashibara, 2023-02-05:
gazebo上で構築したシミュレータ環境のノードの繋がり

[image]

yasuohayashibara, 2023-02-05:
GNSSをメインで使用するときにはこちらがおそらく正解

[image]

yasuohayashibara, 2023-02-05:
#70 にシミュレータで検証した結果を載せました．

yasuohayashibara, 2023-02-14:
実ロボットレベルでまだ問題は解決していませんが，開発はとりあえず終了したので，こちらのissueは閉じます．

closed 2023-02-14 by yasuohayashibara