open-rdc/orne-box #71: ORNE-box3の検討

yasuohayashibara, 2023-02-14:
ORNE-box3の検討を行う．

yasuohayashibara, 2023-02-14:
ORNE-boxとの比較

ORNE-box　13,140g　W518 x D585 x H304
[image]

ORNE-box3（現在の案）　16,840g　W516 x D605 x H317
[image]

yasuohayashibara, 2023-02-14:
ONRE-boxからORNE-box3への変更

メリット
１）堅牢な防水対策
２）ケーブルを前後に出せる
３）ボックスの容積が大きい
４）ミスミですべて加工できる

気になる点
５）車高が高い
６）前のアームの固定方法

yasuohayashibara, 2023-02-14:
車高が高くなっていますが，意図があれば教えてください．

ORNE-box
[image]

ORNE-box3
[image]

yasuohayashibara, 2023-02-14:
車高を高くした件と関連しそうな気がしますが，前輪キャスタが一部アルミフレームで固定はされていますが，板材で固定するような形式になっています．
これは車高を上げたために，ブラケットで固定するのが難しくなったからでしょうか？

ORNE-box
[image]

ORNE-box3
[image]

yasuohayashibara, 2023-02-14:
LiDARを固定するアームがアルミフレームでは無く，板に固定されていますが，これは矢印の部分のネジが干渉するためにずらしているという認識で構いませんでしょうか．

[image]

yasuohayashibara, 2023-02-14:
先走っているかもしれませんが，以下の接続方法は部品を取り外しても防水が保たれるという面で非常に良いと思いますので，可能であれば前のアーム部にも適用したほうが良いかなと考えています．
取り付けの自由度も生まれますので，多目的で使用するというロボットに向いていると思います．

あと，板で支えないということを前提にすると，3mmは少々厚いですが，これはミスミの加工の制約によるものですよね？

[image]

maHidaka, 2023-02-14:
> 車高が高くなっていますが，意図があれば教えてください． 車高が高いと底面と地面が接触するおそれは低くなりますが，傾いたときに転倒しやすくなります．
> 
> ORNE-box [image]
> 
> ORNE-box3 [image]

orne-boxは上面に後輪がはみ出していて、上面パネルを取り外す際に干渉して非常に不便という要望がありました。加えて台車型の構想として、上面に上物を載せる際にも制約になっていたため、今回は上面をフラットになるように変更しています。
底板の高さは、モータ周りの剛性を確保したかったので上面を高くしたことに伴って上がっています。重心位置の問題ですが、車軸より上にはあるものの、フットプリントが大きいこともあって転倒の恐れはないかなとは考えています。lidarへの振動の影響も許容範囲かなと考えておりますが、どうでしょうか。
[image]

maHidaka, 2023-02-14:
全面パネル部のキャスタとlidarのアームの構造についてですが、これらはどちらも主に防水対策のためにこのような構造に変更しています。
orne-boxでは前、後面パネルにて内側のフレームと外部フレームを直接ブラケットで固定する構造でしたが、内部フレームが露出しているため浸水が防げませんでした。
orne-boxV2では後面パネルにて完全にパネルで内外を分離する構造を採用し、一定の防水効果が実証できました。
また、内側のフレームの後入れナットと外部フレームをパネルのバカ穴を介して挟むように締結することで、パネルと内部フレームの固定を兼ねるようになっていました。その構造だと後入れナットの位置調整が難しく組み立て難易度が高い問題や、アームやハンドルの取り外し時に毎回パネルまで外すことになるという問題点が発生します。
orne-boxV3ではそれらの問題点を踏まえて、パネルで内部を密閉し、パネルと内部・外部フレームの締結部分を分けて分解時にも本体の防水構造は担保されるようにしてみたという経緯になっています。
また、orne-boxではキャスタのアームはブラケットによって摩擦で固定されているだけなので、走行振動や分解組み立てによって位置がずれ、機体が傾くという問題がありましたが、側面パネルのねじ穴に固定することで(ねじ位置による担保なので微妙ですが)改善を試みています。
この構造のデメリットとしてはネジ部M5x8本で剪断方向に荷重を受けていることで、剛性的にはベストではない点が挙げられます。不足していそうならアーム間に追加でフレームをブラケットで固定しようかなと考えています。

yasuohayashibara, 2023-02-15:
> orne-boxは上面に後輪がはみ出していて、上面パネルを取り外す際に干渉して非常に不便という要望がありました。
> 加えて台車型の構想として、上面に上物を載せる際にも制約になっていたため、今回は上面をフラットになるように変更しています。

こちらORNE-boxからORNE-box3のメリットに加えたほうが良い内容でした．
付け加えました．賛成します．

> 底板の高さは、モータ周りの剛性を確保したかったので上面を高くしたことに伴って上がっています。重心位置の問題ですが、車軸より上にはあるものの、フットプリントが大きいこともあって転倒の恐れはないかなとは考えています。lidarへの振動の影響も許容範囲かなと考えておりますが、どうでしょうか。

ここらへんぼんやりしたことしか言えない状況ですが，将来的にアーム付きの上半身を取り付けて凸凹のある道を走行させることも考えていますので，重いバッテリはできるだけ下に置きたいというのはあります．
低重心の台車があると上半身の設計の制約が減りますので，デメリットがさほど無ければ下げていただいた方が汎用性が高まるかなと思います．

> 全面パネル部のキャスタとlidarのアームの構造についてですが、これらはどちらも主に防水対策のためにこのような構造に変更しています。
> orne-boxでは前、後面パネルにて内側のフレームと外部フレームを直接ブラケットで固定する構造でしたが、内部フレームが露出しているため浸水が防げませんでした。
> orne-boxV2では後面パネルにて完全にパネルで内外を分離する構造を採用し、一定の防水効果が実証できました。

こちらはメリット「１）堅牢な防水対策」に入れましたが，防水効果を高める処置なので賛成します．

> また、内側のフレームの後入れナットと外部フレームをパネルのバカ穴を介して挟むように締結することで、パネルと内部フレームの固定を兼ねるようになっていました。その構造だと後入れナットの位置調整が難しく組み立て難易度が高い問題や、アームやハンドルの取り外し時に毎回パネルまで外すことになるという問題点が発生します。 [...]

こちらの問題を把握しました．
解決策を考えてみます．
できるだけ本日中に案を出すようにします．

yasuohayashibara, 2023-02-16:
時間がかかっており，申し訳ありません．
本日は夕方まで拘束されるので，現状をお伝えします．

現在，下図のようなアルミフレームを固定して，それに前輪とセンサーをマウントする方法を検討しています．

[image]

ブラインドジョイントを使用することで，ネジとの干渉が無くなるので，好きな位置にマウントできるようになります．

以下の部品，ただしネジの長さを変更する．
https://jp.misumi-ec.com/vona2/detail/110300464650/?HissuCode=HCJ5

[image]

問題は，ネジの長さ的に11.5mmが望ましいのですが，それが見つからないことです．
以下は12mmのネジ（BOX-BCB5-12）を入れたものですが，アルミ材に干渉してしまいます．

[image]

あと，ナットの位置がずれるという問題ですが，一つは長ナットを入れる方法があります．
相対的な位置が固定されるので，良いと思いますが，重くなるおそれがあるので，重量を見てから判断しようとしています．
[image]

maHidaka, 2023-02-17:
@yasuohayashibara 忘れていたのですが、meviyの板金部品では穴位置にナットを取り付けられるので、板材のタップ部分を圧入ナットに変えると板側の強度はあまり気にならないようになるかなと思いました。圧入は強度について公称値としての保証がどうもないようなのですが、ラジアスとアキシャルどちらも荷重を面で受けられるのでタップを切る問題点はある程度改善すると思います。

ナット位置の問題ですが、L寸指定の長ナットは価格が高いのと、少し重くなるということで保留していました。
イモネジで固定できるロックナットは、組み立て時にナットと板材の仮固定と本固定を行う必要があるので、多少組み立て難易度が上がるかなと考えていました。
しかし、通常のナットと価格がさほど変わらないことと、やはり毎回分解のたびにパッキン越しに穴位置を調整するデメリットは大きいので、採用したいなと思います。

yasuohayashibara, 2023-02-19:
@maHidaka 採用するしないは別として，午前中に案を提示できるように進めていたのですが，ミスミのサイトがメンテナンスに入って作業ができなくなりました．
案の提示は明日以降になります．
待てないようであれば，現在の案で進めていただいて構いません．

現在の状態
[image]

ただし，一点気になる点を新たに見つけました．
以下の前輪の固定部ですが，せん断力が矢印のネジに加わるような構造になっています．
ORNE-boxは加わらない訳では無いですが，できるだけ分散させるような構造にしています．
ご参考まで．

ORNE-box3
[image]

ORNE-box
[image]

yasuohayashibara, 2023-02-20:
完全ではないですが，とりあえず作れそうなところまでは案を作成しました．
再度になりますが，あくまでオプションと考えていただいて構いません．
最終的にどうするかはお任せします．

[image]

以下変更点です．

１）底面（重心）を下げました．

提案
[image]

原案
[image]

２）前輪のアームの固定をアルミフレームを繋ぐ形式に変更（板材を挟んでですが）
また，赤矢印のボルトのせん断力を緩和する構造に変更

[image]

３）URGもアルミフレームからつなぐ形に変更
少し取り付け位置を変更できるようにしたかったが，あまり変わりなし．
取り外しても防水が保たれるメリットはある．

[image]

懸念点

yasuohayashibara, 2023-02-24:
今日の議論で使わない方針となりましたが，一応CADデータのリンクを送ります．

https://drive.google.com/drive/folders/1UxXHZjSLZbgVddQ1Xgc8diDWUWhWv10a?usp=sharing

yasuohayashibara, 2023-03-09:
議論が収束したのと，長くなったのでissueを閉じます．

closed 2023-03-09 by yasuohayashibara