open-rdc/orne-box #74: IMUのキャリブレーションについて

Mochizuki12, 2023-06-24:
静止時にIMUのＺ軸回転のドリフトが原因でロボットの姿勢が回転する問題があったため，ADI_IMU_TR_Driver_ROS1のキャリブレーション機能を使用してキャリブレーションを行う．

過去120秒のデータを使用してキャリブレーションを行うため，120秒間静止させてからserviceをコールするとキャリブレーションを行える．以下に現在のキャリブレーションを行う手順を示す．

１．script_orneの中にある, imu_calibration.shを実行．
２．120秒のカウントが始まるのでその間ロボットを動かさない．
３．calibration doneとの表示が出たらキャリブレーション完了．

以上の手順でキャリブレーションを行うことができる．
**ただし，box2の電源を落とすとキャリブレーションがリセットされるので，電源を入れたら毎回おこなう必要がある．**

IMUのROS 1パッケージ
https://github.com/technoroad/ADI_IMU_TR_Driver_ROS1

コールするservice
```
$ rosservice call /imu/cmd_srv "cmd: 'START_BIAS_CORRECTION'
args:
- ''"
```

Mochizuki12, 2023-06-24:
また，adi_driverのキャリブレーション機能を使用した結果を以下示す．現在のbox2のjetsonの環境ではadi_driverは使用できなかった．他のPCでは動作確認済み（melodic, noetic）．

adi_driverは過去40秒のデータを使用してキャリブレーションを行う．40秒間静止させてからserviceをコールするとキャリブレーションを行える．
**ただし，box2の電源を落とすとキャリブレーションがリセットされるので，電源を入れたら毎回おこなう必要がある．**

以下に実行した動画のURLを添付する．
https://www.youtube.com/watch?v=ont-yWjL6X0

使用したROS 1パッケージ
https://github.com/tork-a/adi_driver

コールするservice
```
rosservice call /imu/bias_estimate
```

Mochizuki12, 2023-11-15:
IMUのキャリブレーションの効果を18階を1周した時のオドメトリとIMUで補正したオドメトリを表示して確認した．
現在のシステムでは，オドメトリのデータは，オドメトリの生データである odom とIMUのデータでオドメトリを補正した　odometry_filteredの2種類がある．

1．odom
[image]

2．キャリブレーション前のodometry_filtered
[image]

3．キャリブレーション後のodometry_filtered
[image]

キャリブレーション前は閉じていなかったが，キャリブレーション後はループが閉じているため，キャリブレーションの効果があることがわかった．

yasuohayashibara, 2023-11-20:
@Mochizuki12 Yuya Mochizuki
つくばチャレンジも終了したのでissueを閉じてください．

closed 2023-11-20 by Mochizuki12

yasuohayashibara, 2023-11-20:
以下，データとして追加しておく．
つくばチャレンジでのオドメトリの値

[image]

ちなみに，IMUを利用していないORNE-box3は以下のような感じ

[image]