open-rdc/orne-box #96: robot_localizationのodometry/gpsトピックの値がおそらく正しくない

hellocit, 2023-12-28:
robot_localizationパッケージに含まれるnavsat_transform_nodeのodometry/gpsトピックの値がおそらく正しくない

hellocit, 2023-12-28:
以下がrostopic echoの出力である．クオータニオンのwの値だけが1である．

`---
header:
  seq: 1263
  stamp:
    secs: 1703420906
    nsecs: 628080368
  frame_id: "odom"
child_frame_id: ''
pose:
  pose:
    position:
      x: 0.0
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  covariance: [0.017689000000000003, -7.59710849412839e-21, 0.0, 0.0, 0.0, 0.0, 6.565395813338196e-21, 0.017689000000000003, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.28302400000000005, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---`

hellocit, 2023-12-28:
原因として考えられることを以下に示す．
- 渡すべきトピックが渡されていない
- TFのリンクが適切でない
- 適切なパラメータ設定でない

hellocit, 2023-12-28:
渡すべきトピックについて調査を行う．
[navsat_transform_node]()のwikiによれば，サブスクライブするトピックは以下の3個である．
- imu/data
- odometry/filtered
- gps/fix
どのトピックにも中身があることとおそらく不正な値でないことを確認した．また自分のPC側で/fixから/gps/fixにremapしたので名前にも問題がないと考える．

hellocit, 2023-12-28:
TFのリンクについて調査した結果をまとめる．orne_box.launchでxacroやurdfを読み込んでいる．また，センサ類はそれぞれのlaunchファイル内でconfigを読み込んでいる．このことから分散実行でGNSSを起動するときは子機側でGNSSの起動を行う必要がありそうである．

現状のbox2でGNSSを使う時のlaunchファイルは以下のものを使用する．またその簡単な説明を行う．

- ナビゲーションを除くノード類の起動
[orne_box.launch](https://github.com/open-rdc/orne-box/blob/TC_2023/orne_box_bringup/launch/orne_box.launch)

- GNSSレシーバの起動
[ublox_device.launch](https://github.com/open-rdc/orne-box/blob/TC_2023/orne_box_bringup/launch/includes/ublox_device.launch)

ublow_device.launchによってGNSSレシーバは起動するが現在の構成では，TFのリンクを定義しない起動コマンドを用いている．従って，TFのリンクが適切でないという結論でbox2内で真偽を調査する．

hellocit, 2023-12-28:
パラメータについて
おそらくこの[ファイル](https://github.com/open-rdc/orne-box/blob/TC_2023/orne_box_bringup/launch/includes/robot_localization.launch)を見る必要がありそう．

yasuohayashibara, 2023-12-28:
robot_localizationにGNSSを含めないようにしてください．
理由は研究ノートを参照してください．
初回に１回だけ参照していたような気もしますので，その場合は問題ないかもしれませんが，別々に取得したほうが比較しやすい気がします．

以下のノードを使用していると思いますが，まずは正しい値が出力されているかを確認してください．
https://github.com/KumarRobotics/ublox

前に計測した様子が報告されていましたが，あれは上記を使用したものでしょうか．
そうであれば，後は計測して解析するという段階に来ていると思います．

hellocit, 2023-12-28:
申し訳ありません．私の環境の情報と経緯が不足していました．[このissueに](https://github.com/open-rdc/orne-box/issues/29#issuecomment-894268924)よればbox2に積まれているjetsonで[このGNSSレシーバ](https://www.bizstation.jp/ja/drogger/dg-pro1rw_index.html)は使用できません．ですのでjetsonを親機，自分のPCを子機としてROSの分散実行を行っています．GNSSレシーバを起動する時は，[このパッケージ](https://github.com/terakenxx/f9p_ichimill)を使用して
`roslaunch f9p_ichimill gps_ichimill.launch`
上記のコマンドを子機で実行し，GNSSを起動しています．その時はこの[コンフィグファイル](https://github.com/open-rdc/orne-box/blob/TC_2023/orne_box_bringup/config/ublox/zed_f9p.yaml)を読み込んでいないのでTFが適切に設定されていないだろうと予想しました．

hellocit, 2023-12-28:
> robot_localizationにGNSSを含めないようにしてください．
理由は研究ノートを参照してください．
初回に１回だけ参照していたような気もしますので，その場合は問題ないかもしれませんが，別々に取得したほうが比較しやすい気がします．
承知しました．

> 以下のノードを使用していると思いますが，まずは正しい値が出力されているかを確認してください．
別のROSパッケージを利用しています．正しい値かどうかに関しては研究ノートに示したとおり問題ないと考えます．

yasuohayashibara, 2023-12-28:
そうであれば，特に問題ないかもしれません．
以下が計測できていれば，次の解析のステップに進めます．

１）従来の自己位置推定の結果
２）GNSSの計測結果

念のため`rqt_graph`でGNSSの出力がどこにも接続されていないことを確認しておくと良いと思います．
rosbagを取得する場合はrosbagには接続しているような頭になるかもしれませんが．

hellocit, 2023-12-28:
承知しました．ありがとうございます．おそらく1)と2)は取得できていそうなので確認します．これからの内容はorne-boxとは直接かかわらないのでissueを閉じます．

closed 2023-12-28 by hellocit