open-rdc/orne-box #97: TFにおけるリンク間の差を求めたい

hellocit, 2024-01-05:
base_linkのxとy座標の値-gnss_linkのxとy座標の値

上記の計算を行いたい．計算したい対象の画像とtfのツリーを以下に示す．
[image]

[image]

hellocit, 2024-01-05:
[chatgptに作らせたスクリプト](https://github.com/hellocit/GNSS-research/blob/main/test/test.py)で解決している可能性もある．

yasuohayashibara, 2024-01-05:
TFを用いた座標の変換ですが，上記のスクリプトのような処理でできると思います．
念のため，手計算で解いた座標と上記の計算結果が一致しているかどうかを確認してから進めたほうが良いと思います．
（間違った方法が提示されているおそれもありますので．）

hellocit, 2024-01-12:
プリント分を用いたところ値が出力されました。

hellocit, 2024-01-12:
<xacro:sensor_gnss name="gnss" parent="base_link">
    <origin xyz="-0.12 0.12 0.9" rpy="0 0 0"/>
  </xacro:sensor_gnss>

上記の値と一致していたので問題はないと考えます。

hellocit, 2024-01-12:
あとはgnssの値をロボットのいる座標系に変換できればよいのか...

hellocit, 2024-01-12:
https://github.com/open-rdc/orne-box/blob/39cc2e3171b137bfcf8c1c9087dae9fa7548f1ae/orne_box_description/urdf/orne_box_3d_lidar.urdf.xacro#L79C1-L82C23

xとy座標は上記の値と一致していたので差を求める計算には問題がないと考えます。

yasuohayashibara, 2024-01-12:
解決したらissurを閉じていただいて構いません．

hellocit, 2024-01-12:
承知しました．ありがとうございます．

closed 2024-01-12 by hellocit