open-rdc/orne-box #105: ekf_nodeにimuのデータが含まれていない

kyo0221, 2024-09-26:
ekf_filter_nodeに/odomと/imu/dataを渡してセンサフュージョンすることを想定しているが, ekf_filter_nodeの出力である/odometry/filteredには/imuの値が適応されておらず/odomと同じ値が出てしまっている.

白 : /odom
赤 : /odometry/filtered
[image]

```
ros2 topic info /odom
```
```
ros2 topic info /imu/data
```
でekf_filter_nodeがサブスクライブしていることは確認している.

ekf_filter_nodeに使われているekf.yamlの記述に問題があると思われる.
https://github.com/TakumiIshiguro/orne-box/blob/bag-fix/orne_box_bringup/config/robot_localization/ekf.yaml

kyo0221, 2024-09-27:
ekf_filter_nodeのパラメータについて書いてあるドキュメント
https://docs.ros.org/en/melodic/api/robot_localization/html/state_estimation_nodes.html#ekf-localization-node

kyo0221, 2024-10-01:
/odometry/filteredがIMUのヨー角の情報で動くようになった.
車輪を浮かせた状態で車輪を回したら/odomだけが回転することを確認した.

IMUが読み込めていなかったのはtfの問題のようだった.

白：/odom
赤：/odometry/filtered
[image]

現状では/odometry/filteredの向きと進行方向が合っていない動きをしているので原因を探っている.

kyo0221, 2024-10-02:
box2のtf
[image]
[image]

box3のtf
[image]
[image]

TakumiIshiguro, 2024-10-02:
ac311e8e938fca0dcadc953cdfa839a17b845609
frame_idを変更した

変更後のrqt_tf_treeは以下のように変化した
[image]

再び/odometry/filteredを確認すると/odomよりもきれいな軌跡が描けた
imuのキャリブレーションは行っていない
[image]

kyo0221, 2024-10-06:
津田沼キャンパスの外周部分を1週させた/odometry/filteredの軌跡です.
IMUはキャリブレーション済みになっています.

[image]

ekf_nodeでimuのデータを読み込めるようになったのでcloseします.

closed 2024-10-06 by kyo0221