open-rdc/orne-box #106: つくばの実験走行会(10/06)におけるORNE-box2の故障

yasuohayashibara, 2024-10-07:
10/06のORNE-box2が故障した可能性がある．
ハードウェア的に変更したことや気になったこと，故障の原因と思われることがあれば情報共有をお願いします．

yasuohayashibara, 2024-10-07:
ハードウェアに関しては、ディスプレイのbox3へ付け替え､鉛蓄電池を交換をしていただきました。

osukeishihara, 2024-10-07:
故障の原因の考えられること
・鉛蓄電池の交換
・box2内のディスプレイの取り出し(2回)

また、モーターが動かなくなる直前は、「点字ブロックを横切り、経路から外れそうになり、緊急停止」を繰り返しており、その際にモーターが動かなくなった。
点字ブロックでのロボットの揺れも原因ではないかと考えられる

Nomura1024, 2024-10-07:
@osukeishihara 
緊急停止スイッチを押したとき空転させたりした記憶はある？

osukeishihara, 2024-10-07:
@Nomura1024 
空転とは具体的にどのようなことでしょうか？

Nomura1024, 2024-10-07:
@osukeishihara 
緊急停止スイッチ押した時にモータの動力ではなく手押しで動かしたり，ロボットを自分側に引っ張って急激にモータを逆回転をさせたりとか．

osukeishihara, 2024-10-07:
@Nomura1024 
点字ブロックを横切り、経路から外れ、緊急停止ボタンを押した後は、元の経路に戻すために手押しで複数回戻していました。

osukeishihara, 2024-10-07:
@Nomura1024 
また、電池の消耗が激しくコースの途中で電源を落として、手押しで帰るということが2.3回ありました。

Nomura1024, 2024-10-07:
その時は，モータのコネクタ外した？

yasuohayashibara, 2024-10-07:
以下にモータドライバの故障が疑われる時のトラブルシューティングがあります．

https://github.com/open-rdc/orne_navigation/wiki/t_frog%E3%83%A2%E3%83%BC%E3%82%BF%E3%83%89%E3%83%A9%E3%82%A4%E3%83%90_TF_2MD3_R6_%E6%95%85%E9%9A%9C%E8%A8%BA%E6%96%AD

shin0410, 2024-10-07:
確認した内容についてですが, 当日teleopの操作から/cmd_velがパブリッシュされていることは確認しました. それ以外のコネクタの繋ぎ直しや繋ぎ変えなどの確認はおこなっていません.

Nomura1024, 2024-10-14:
10/11に行った[故障チェック](https://docs.google.com/document/d/1OR8RXi-7EK7Q_FC_9wdiQdopKhK6nFCTAJwTAPcjkmE/edit?tab=t.0)

Nomura1024, 2024-10-14:
[トラブルシューティングでは](https://github.com/open-rdc/orne_navigation/wiki/t_frog%E3%83%A2%E3%83%BC%E3%82%BF%E3%83%89%E3%83%A9%E3%82%A4%E3%83%90_TF_2MD3_R6_%E6%95%85%E9%9A%9C%E8%A8%BA%E6%96%AD)ではLED5が点灯しているとMOSFETやゲートドライバの故障とあるが[T-flog](https://t-frog.com/products/motor_driver/troubleshooting.php)ではLED5が消灯している時にMOSFETやゲートドライバの故障であると書かれている．

Nomura1024, 2024-10-14:
そのため[前回](https://github.com/open-rdc/orne-box/issues/106#issuecomment-2409562214)チェックしていたときに消灯していたためMOSFETやゲートドライバの故障だと断定できる．
過去に先輩(井口先輩）からドライブ基板（FETの基板)のみ予備があると教えていただいていため交換を行いコントローラーで動かせることを確認した

yasuohayashibara, 2024-10-15:
修理依頼しますので，モータドライバ基板を下さい．

shin0410, 2024-10-15:
> 10/11に行った[故障チェック](https://docs.google.com/document/d/1OR8RXi-7EK7Q_FC_9wdiQdopKhK6nFCTAJwTAPcjkmE/edit?tab=t.0)

実施手順
①bringupで動作を確認する
rosトピックの cmd_velはでている
rosトピックの odomはでている
緊急停止ボタンを押した際にモーターは脱力する
テスターでモータードライバに24Vかかっているか確認した

②モータードライバをbox2から取り外し安定化電源に接続し確認する
外観に問題は見られなかった
電源基板に24V, con1端子に5V, con13端子に24Vをモータードライバ基板に入力する
コントロール基盤led4の点灯とドライブ基板裏側のled5が点灯していないことを確認した

③box2にモータードライバ基板を戻す
再度led4の点灯、led5消灯を確認した
jetsonを起動しbringupした後, led3が点灯し通信はできていると考えられる

④緊急停止の基板も確認した, ボタンにあわせて動作していることを確認した

→box2のモータードライバ(特に下段のドライブ基盤)が原因と考えられる

Nomura1024, 2024-10-16:
故障の状況を伝えるドキュメント
https://docs.google.com/document/d/1aVs_W160C4Z3eAbh-6ra9GswekjgRNUs2qAu8AbvnAo/edit?usp=sharing

shin0410, 2024-10-25:
[Uploading 2024_10_25orne-box構成の変更.pptx…]()

yasuohayashibara, 2024-10-26:
@Nomura1024 @shin0410 
ファイルはリンクではなく実体を貼り付けるようにしてください．
リンク先のファイルはしばしば消えてなくなるので，こちらに実体を残したほうが良いです．

shin0410, 2024-10-26:
[10_6実験走行故障記録.pdf](https://github.com/user-attachments/files/17532218/10_6.pdf)
[2024_10_25orne-box構成の変更.pdf](https://github.com/user-attachments/files/17532219/2024_10_25orne-box.pdf)

closed 2025-04-13 by Ryusei-Baba