open-rdc/orne-box #107: 鉛蓄電池の状態について

Nomura1024, 2024-10-16:
研究室内にある鉛蓄電池の個数と状態を[スプレッドシート](https://docs.google.com/spreadsheets/d/14IJ6X2vBv4dSSqJYziys2-XczPVzkVwIGgYs5kDiRbY/edit?gid=0#gid=0)記録する

Nomura1024, 2024-10-16:
研究室内にある鉛蓄電池の個数は12個であり，全て鉛蓄電池にIDを付けた
[image]

Nomura1024, 2024-10-16:
鉛蓄電池の状態を計測するために[バッテリーチェッカー](https://amzn.asia/d/3DKVvdr)を用いる．
問題点
- 鉛蓄電池自体のCCA等の情報がないため劣化具合が不明(鉛蓄電池が車用でないため情報がない）
- IRのみでの劣化の判断は難しい

Nomura1024, 2024-10-18:
鉛蓄電池の状態をみるのに[電子負荷テスター](https://www.amazon.co.jp/dp/B0CT5LG947/ref=twister_B0CT54QM4W?_encoding=UTF8&psc=1)を用いるのがよさそう

yasuohayashibara, 2024-10-18:
抵抗値可変の大型の抵抗器がありますので，それが利用できる可能性があります．
探しても型番が出てきませんが，新習志野にあります．

kokiikeda6, 2024-11-21:
新習志野にあった負荷装置で鉛蓄電池の状態確認を行います.
使用する負荷装置：maynuo DC Electronic Load M9711 ([リンク](https://www.maynuo.eu/product/1188811/maynuo-m9711-dc-electronic-load-0-30a-0-150-v-150-w))
測定する鉛蓄電池：12V14Ah WP14-12SE ([リンク](https://akizukidenshi.com/catalog/g/g112641/))

確認方法
 一定電流を放電させ続け, 一定以下の電圧になるまでの時間を計測

確認手順
1. 鉛蓄電池を満タンまで充電
2. バッテリーを放電, 時間を計測
　a. 負荷装置の設定, 定電流モード(CC) 2.38 [A]
　　i. 電源オン | [POWER]
　　ii. 流す電流設定 | [I-set] → [数字](流す電流設定, 今回は2.38 [V]) → [Enter]
　　iii. 開始時に必要な電圧, 終了時の電圧を設定 | [Shift] → [1] → [数字](ON SETの電圧設定, 今回は12 [V]) → [Enter] → [数字](OFF SETの電圧設定, 今回は11 [V]) → [Enter]
　　iv. 計測開始, 終了 | [On/Off]
　b. 負荷装置に表示される電圧を監視
　c. 11.0 [V] になるまでの時間を計測 (2.38 [A] で放電時, 終止電圧 10.2 [V])
3. SOH [%] (健康度) を計算 (満タン時の電圧 [V] → 11 [V] のときの時間で計算)
　a. 今のバッテリーで11[V]になるまでの放電時間 = 計測した時間 [h]
　b. 新品のバッテリーで11[V]になるまでの放電時間 = 4.5 [h]
　c. SOH [%] = 2.38 [A] * 計測時間 [h] / 2.38[A] * 4.5 [h]
　d. SOH [%] = 計測した時間 [h] / 4.5 [h]

kokiikeda6, 2024-11-29:
@yasuohayashibara 
研究室にあった11個の鉛蓄電池(WP14-12SE)を調査しました. ([記録したスプレッドシート](https://docs.google.com/spreadsheets/d/14IJ6X2vBv4dSSqJYziys2-XczPVzkVwIGgYs5kDiRbY/edit?gid=0#gid=0))
使用できると思われる鉛蓄電池は6個で, 現在予備がない状況なので, SOHが23％, 57％, 59％, 44％のものを廃棄し, 新たに4つの鉛蓄電池を購入していただきたいです.
amazon
https://www.amazon.co.jp/LONG-12V-14Ah-%E9%AB%98%E6%80%A7%E8%83%BD%E3%82%B7%E3%83%BC%E3%83%AB%E3%83%89%E3%83%90%E3%83%83%E3%83%86%E3%83%AA%E3%83%BC%E3%80%90%E9%AB%98%E8%80%90%E4%B9%85%E3%82%BF%E3%82%A4%E3%83%97%E3%80%91-WP14-12SE/dp/B01KQ6KI5Y

kokiikeda6, 2024-11-29:
廃棄の方法については, 経験のある中村さんに教えていただきました. 一号館二階で手続きができるようなので, 来週中には手続き行い, 廃棄します.

yasuohayashibara, 2024-11-29:
4つ購入しました．
以下お願いします．

・電池の廃棄
・電池の管理　購入日が分かるようにしてください．
・電池の使い方の講習　過放電すると利用できなくなることに関して，周知するようにしてください．
　たしか充電しっぱなしでも放電することがあったような気がしますので，それも共有して下さい．

kokiikeda6, 2024-12-04:
劣化の進んだ鉛蓄電池(4つ)を廃棄しました.
1号館2階の施設担当のところで廃棄の予約をし, 指定の日時に8号館裏のゴミ捨て場(⑤あたり)に持っていく形でした.
[image]

yasuohayashibara, 2024-12-04:
リマインドしなくても実施してくれたのですね．
ありがとうございます．

池田君でなくても構いませんが，引き続き電池の管理方法に関してのルールづくりと周知をお願いします．
劣化した電池を使わざるを得ない状況にならないようにみなさんで協力していただければと思います．

osukeishihara, 2025-04-25:
https://docs.google.com/document/d/1D32dtyhCZ_GC9Ntw0HDUphjFQzlFM4zJyWnV_mvu3ik/edit?usp=sharing
鉛蓄電池の管理マニュアルを作成しました.

closed 2025-04-25 by Ryusei-Baba