open-rdc/orne-box #119: つくばチャレンジ2024で発見した問題と改善方法（box3）

Ryusei-Baba, 2024-12-09:
タイトル通り、TC2024で発見した問題とその改善方法を記述する。

1. Jetpack6.0でジョイコンが使用できない問題
→https://github.com/open-rdc/orne-box/issues/116

2. rfansのscanデータが揺れる問題
→https://github.com/open-rdc/orne-box/issues/113

3. ekf_nodeにimuのデータが含まれていない
→https://github.com/open-rdc/orne-box/issues/105

4. ros2のシステムで3次元自己位置推定の実装
→未解決
　https://github.com/open-rdc/orne-box/issues/104

5. ロボットが直進しない問題
→https://github.com/open-rdc/orne-box/issues/103

6. 起動の簡略化(ROS2) 
→https://github.com/open-rdc/orne-box/issues/101

7. ros2のIMUキャリブレーション
→https://github.com/open-rdc/orne-box/issues/100

8. ナビゲーション中のCPU使用率が高い（CPU使用率90％以上）
→未解決
（RVizの表示やrosbagのTopicを絞って対策したが、本走行にて90％以上CPUを使用していた。本走行では、ナビゲーション・rosbag・信号認識を起動していた）

9. Jetson Orin NXが急に落ちる問題
→未解決
　TC2023でも起きた問題で、しばらくの間この問題は起きていなかったが、本走行前日と本走行で頻発した。昨年度まではバッテリの電圧（鉛蓄電池）が問題ではないかと考えていたが、今年に関してはバッテリの電圧（鉛蓄電池）に余裕があるにも関わらずJetsonが落ちた。また、ポータブル電源でも同様の症状が見られた。
　症状は異なるかもしれないが、以下のリンクのようにJetsonには磁気や磁力による脆弱性があるのかもしれない。つまり、ORNE-box3のJetsonに搭載しているキャリアボード（A603 Carrier Board for Jetson Orin NX/Nano）に磁気や磁力が悪影響を起こしているのかもしれない。
https://x.com/ystk_hara/status/1864869592386146763?s=46

Ryusei-Baba, 2024-12-11:
CPU使用率が高い問題
Jetsonのパフォーマンス最大化を行ったところCPU使用率が10％ほど低下した。
https://github.com/open-rdc/orne-box/issues/121

Shuma-s58, 2024-12-20:
遅くなりましたが, 上記の**8. ナビゲーション中のCPU使用率が高い**において言及されている信号認識のリポジトリは以下になります.

[信号認識のリポジトリ](https://github.com/open-rdc/cross_traffic_manager_cpp)