open-rdc/orne-box #123: box1の鉛蓄電池の減りが速い

YuseiShiozawa, 2025-03-12:
box1の鉛蓄電池の電圧の減り具合がやたら速いです。津田沼チャレンジ2/7くらいまでは起動時25ｖ程度で走れていましたが、3/10には起動時23ｖ程度になっていました。充電器は満タンの表示でした。(box1に関して、2/7-3/10まで少なくともb3は使用していないです)

また、電圧計で測ってみると、片方14ｖ(ID: 0), もう一方は9ｖ(ID: 3)になっていました。このことから、充電器に満タンと表示されている状態で、片方のみ電圧が低いのはハードウェア側に問題があるのではないかと考えています。そして、原因が分かるまでは電池を交換して使用するのは控えたほうがよいと考えています。

原因、解決方法が分かる方がいましたら教えていただきたいです。
また、他に状況を知っている方（2/7-3/10にbox1を使用した方等）がいましたら教えていただきたいです。

yasuohayashibara, 2025-03-12:
電池に関しては，どこかで議論されてこちらにまとまっています．
石原君が経緯を知っていると思います．

https://docs.google.com/document/d/1D32dtyhCZ_GC9Ntw0HDUphjFQzlFM4zJyWnV_mvu3ik/edit?usp=sharing

YuseiShiozawa, 2025-03-13:
ドキュメントの通り鉛蓄電池の状態を確認したところ、SOHは110% → 93%となりました。
また、バッテリー充電器を用いてID: 3の鉛蓄電池を個別で充電したところ、起動時の電圧は25.9ｖ（正常な値）でした。

box1は鉛蓄電池を個別で充電するようにした方が良いのかもしれません。

yasuohayashibara, 2025-03-13:
再度になりますが，どなたかサポートに入ってください．

Ryusei-Baba, 2025-05-13:
こちら、引き続き同様の状況が発生しております。

バッテリーチャージャー接続時には27V以上が表示されますが、チャージャーを切断すると電圧が一気に23Vまで低下します。
このことから、電気的（回路的）な故障が発生している可能性が高いと考えられます。

[image]
[image]
[image]

Ryusei-Baba, 2025-05-13:
とりあえずバッテリーチャージャーの方を変えて充電してみたいと思います。

Ryusei-Baba, 2025-05-13:
上記の画像にあるバッテリーチャージャーを変更しても充電できませんでした。
ORNE-boxの内部回路が他のbox2やbox3と違うか故障の可能性があります。

Ryusei-Baba, 2025-05-13:
[このリンク](https://github.com/open-rdc/orne-box/issues/123#issuecomment-2719892934)にあるとおり鉛蓄電池を個別に充電すれば使えます。

yasuohayashibara, 2025-05-13:
こちらの件，野村君の意見を聞きながら，誰かが担当して解決したほうが良いと思います．
バッテリーの出し入れを頻繁にすると，操作ミス，配線ミス，配線の抜き差しによる劣化などで故障する割合が高まりますので，対処したほうが良いと思います．

Ryusei-Baba, 2025-05-19:
@Nomura1024
対処お願いします。
人手が足りなければ手伝います。

onlydcx, 2025-05-24:
@Ryusei-Baba @YuseiShiozawa 
現在原因を調べている途中ですが、box1に設置されていたバランサーが故障している可能性が高いです。

yasuohayashibara, 2025-05-24:
良いですね．
あとで利用するかもしれませんので，できるだけデータや写真を取りながら進めていただければと思います．

onlydcx, 2025-05-29:
実験中に気付いた事ですが、バッテリーの自己放電が異常に早いです。
負荷を接続せずに20時間でそれぞれ、
ID:1 13.19V → 12.87V
ID:2 12.97V → 12.68V
まで放電しています。

yasuohayashibara, 2025-05-29:
ひどいですね．
調査ありがとうございます．
我々固有の問題のような気もしますが，興味深い内容になるかも知れませんので，できるだけデータを取りながら進めたほうが良いと思います．

onlydcx, 2025-05-31:
バランス充電器を新品に交換し1日放置させたところ、電位差が30mV程度になっていたのでバランス充電はできていそうです。
よって、何かしらの原因でヒューズが切れバランス充電ができていなかった事が、片方の電圧が低くなっていた原因なのではないかと思います。またバッテリー自体の劣化も関係していそうに見えます。

とりあえずこのままbox1に組み込んで、様子を見る方針で大丈夫でしょうか？

kyo0221, 2025-12-18:
box1では現状問題になっていないのでcloseします