open-rdc/orne-box #125: 振動で一瞬電源が落ちる問題（box2）

masato4988, 2025-03-22:
コントローラでrosbag recode中に、小さな段差で一時的に電源が落ちる症状が確認された.
症状としては、
・液晶画面が急にブラックアウト
・モータは、コントローラで前進入力をしていても振動の際に止まる
・バッテリ電圧のインジケータが再起動
・jetsonは落ちない
jetsonの電源だけが落ちないのは、無停電電源装置によるものだと思われる

再び同じ段差を画面収録し走行させ、同じ症状が確認された
https://github.com/user-attachments/assets/ba068204-994b-47f2-a9fa-af00cdc0eb5b

Daisuke-joho, 2025-03-22:
https://github.com/user-attachments/assets/41206cb6-edef-41d9-bb6c-16e2a4b588cd
症状が起きた場所
図書館前
正門周辺

masato4988, 2025-03-22:
原因として考えられたこと

box内部を確認したところ、主電源線スイッチの端子に焦げが確認された

また、端子はゆるくなっている
https://github.com/user-attachments/assets/16004ee9-94de-4873-a605-547339145540

振動時に一時的に接触が悪くなり、一時的に停電したと予想する

yasuohayashibara, 2025-03-22:
あと可能性があるのが，USBかなと思います．
USB Type Aは振動に弱く，抜き差しを繰り返すことでより弱くなります．
（グラグラするような状態になっているとしたら，もう利用できないと考えたほうが良いです．）

エラーメッセージを見ると，USBのデバイスが切断されたというものですので，電源かUSBの接続が瞬断しています．
USBは切れると同じデバイス名では再接続はしないので，それ以降使用できなくなります．
（ソフトウェア側で再接続できるようにもできますが，実装が大変なのと，そもそも切れることが問題ですのでお勧めしない方法です．）

つくばチャレンジシンポジウムで話がありましたが，ゴールしたチームはコンピュータ側のUSB Type Aは抜き差し禁止としていました．
USB Type Aで延長したケーブルでUSB Type Cに変換して，そこを抜き差ししていたとのことです．
ケーブルは劣化すれば交換すれば良いので，合理的だと思います．

yasuohayashibara, 2025-03-22:
読み直して気づきましたが，「液晶画面が急にブラックアウト」として回復しないとすると，電源が遮断してそのまま復帰しなかったということでしょうか．
HDMIで接続している場合は，USBではありませんね．
（USBで接続しているとしたら，相変わらずUSB側の問題も疑われますが．．．）

HDMIは電源が復旧すると再度利用できますので，継続的に電源が供給されていないことになります．
そうするとブレーカが落ちたか，継続的にショートしているということになります．
こちらは危険なので，よく原因を追求したほうが良いと思います．

masato4988, 2025-03-23:
液晶画面がブラックアウトすることに関しては、
その後すぐにモニター自体の起動画面が表示され、jetsonからの映像が表示されるようになります。

yasuohayashibara, 2025-03-23:
そうだとしたら，電源の瞬断ですね．

Ryusei-Baba, 2025-04-13:
はんだ付けにより対応済み

closed 2025-04-13 by Ryusei-Baba