open-rdc/orne-box #128: box2の右車輪が, 回転で動作しない／直進時の動き出しが遅い

YuseiShiozawa, 2025-04-13:
box2の右側の車輪に以下のような挙動が見られます

- 低い速度指令（例：回転等）では右車輪が動作しない
- 直進時にも, 右車輪の動き出しが左車輪より遅れる

YuseiShiozawa, 2025-04-13:
#111 
先輩に聞くとここで解決となっている問題らしい. しかし, 同様のパラメータで再発している.

yasuohayashibara, 2025-04-13:
ハードウェア（メカニズム，モータドライバ）に問題があるおそれがあります．
めったに無い良いチャンスですので，原因を追求して，調整の手順に加えたほうが良いと思います．
もし電流が計測できるのであれば，車輪を浮かせた状態で速度制御をして電流値を計測するとハードウェアに問題があるかが分かると思います．
高性能なモータドライバは電流値を計測できる機能もあるので，追加のセンサ無しで状態を判定できるかもしれません．

---

ChatGPTより

減速装置（ギアボックス）やモータユニットの劣化具合，特に摩擦の増加を評価するには，以下のような方法が考えられます．目的に応じて定量的・定性的にアプローチできます．

1．無負荷時の消費電流・電力の測定
方法：モータに負荷をかけずに定速で回転させ，消費電流（または電力）を測定．

判断基準：新品時や正常な状態と比較して，電流値が明らかに高い場合は摩擦トルクが増加している可能性が高いです．

注意点：温度や電圧の影響を排除するために，同条件下で比較してください．

2．トルクセンサによる抵抗トルクの計測
方法：トルクセンサを用いて，出力軸に発生している抵抗トルク（空転時）を直接計測．

応用：ギアの噛み合わせ不良や摩擦抵抗の定量的な測定が可能です．

3．逆起電力（Back EMF）からの診断
方法：モータを手で回し，発生する逆起電圧をオシロスコープ等で測定することで内部の抵抗感を間接的に評価．

所見：スムーズな波形でなければ，軸の偏心や摩擦増加を示唆します．

4．サーモグラフィによる異常発熱の検出
方法：動作中のモータ・ギア部をサーモカメラで撮影し，通常より高温になっている部分を検出．

劣化サイン：潤滑不良や摩耗により異常発熱が起こることがあります．

5．振動解析（加速度センサ）
方法：加速度センサを取り付け，運転中の振動を周波数解析（FFTなど）．

所見：偏摩耗，ベアリング劣化，ギアのバックラッシ異常などが診断可能です．

6．騒音測定（マイクロフォン）
方法：動作音をマイクで収録し，スペクトル解析（音響診断）．

所見：特定の周波数にピークが出るなど異常兆候を捉えられます．

7．機械的インピーダンスの推定（制御応答解析）
方法：モータにステップ入力やサインスイープを加え，出力軸の応答を測定しインピーダンス特性を推定．

利点：モデルベースで劣化度合いのパラメータ化が可能です（摩擦係数，粘性係数など）．

組み合わせの例（おすすめ）：
研究室などで評価する場合：

① 無負荷電流 + ② トルクセンサ + ④ サーモグラフィの組み合わせが実用的かつ再現性があります．

モバイルロボットなど組み込み用途：

① 無負荷電流 + ⑦ インピーダンス解析により定期的な自己診断が可能です．

YuseiShiozawa, 2025-05-08:
このissueを立ち上げてから何故かこの問題が発生しない. 

エンコーダから回転角度を求めてリアルタイムでプロットした動画が[こちら](https://www.youtube.com/watch?v=elqUahePFHs). 
このとき, 実ロボットを前後にコントローラで動かしている. 動画の通り, ずれは左右でズレはまったくないと言える. 

また, かつてこの問題は電池交換により解決されたため, モータ駆動時の電圧が起動時と比べ低い(23.5V)ときに試したが, 同様の問題は発生しなかった.

yasuohayashibara, 2025-05-09:
このような基礎データを継続的に記録していくのはとても良いと思います．
将来的に「真っ直ぐ走らなくなった」などの問題が発生した際に備え，今回のような手順で調査を行うためのマニュアルを整備しておくと効果的です．
それまでは，今回の問題は一旦ペンディングとし，他の作業を優先したほうが良いと思います．

ちなみに，移動中に異音などがしていないかも確認しておいてください．
記憶があいまいですが，以前に車輪の回転時に異音を発するロボットがあったような気がします．
見た目には正常に動いていても，異常が出ているようなケースでは修理対象とした方が良いと思います．

YuseiShiozawa, 2025-05-29:
https://github.com/user-attachments/assets/7d7f3a59-31c6-43f3-9ef4-20202e0b4f5c
再発しました。
こちらの動画は前身、後退、右旋回、左旋回を順に試したものですが、右車輪が動かない/動き出しが遅いことが分かります。

タイヤを外して、シャフトのみで確認しても、同じ動きでした。

YuseiShiozawa, 2025-05-29:
@Nomura1024  
ハードウェア的な知識がまったくないので、対処をお願いしたいです。

YuseiShiozawa, 2025-05-29:
https://github.com/user-attachments/assets/6b96311c-f4a3-45f6-b246-661e625d6db8
手で軽く押してやると動き出します

Daisuke-joho, 2025-05-29:
地面で直進走行させた時は右車輪の動きだしを左車輪の前進方向の力が加わるため右車輪は遅れてですが動き出します.
旋回方向に信号を送ると右車輪に加わる力がないため右車輪を軸とした動きになります.

yasuohayashibara, 2025-05-29:
@YuseiShiozawa 
良いサンプルが得られましたね．
研究の一つの軸として加えることができますので，是非ともデータを取りながら進めてください．
めったに得られない機会ですので，有効に活用したほうが良いです．
まずは，角速度司令に対するステップ応答などを計測した方が良いと思います．
また，本体内に保存されたモータのパラメータを保存したほうが良いと思います．
その他，モータ制御に関するパラメータは全て保存するようにしましょう．
モータに流れる電流値もできれば計測してください．
電流が流れているのにモータが動かないのであれば，そこでハードウェアが問題ということが特定できますので．
最終的にはモータが正常に動作しているかを機械的に判定するところまでできるとみなさんが利用する指針ができそうな気がします．

YuseiShiozawa, 2025-05-30:
@Nomura1024 @onlydcx 
ということで、確認作業に私も参加させていただきたいです。作業日時はslack等で伝えてくださると助かります。

onlydcx, 2025-06-02:
本日試したことをまとめました。

1. モータードライバのポートを左右で入れ替え
     → 右モーターが回らない
2. リレーのケーブルを反転
     → 両モーターが回らない
3. リレー本体を左右交換
     → 両モーターが回らない(継続)
4. リレーとケーブルを左右交換
     → 右モーターが回らない
5. リレー本体を左右交換
     → 右モーターが回らない

ここで初期状態に戻る

6. 右モータをリレーを介さず、モータードライバと直接接続
     → 右モーターが回る
7. リレー本体を左右で交換
     → 両モーター回る

→ リレー回路に問題があるのでは？

リレー の全ての接点が同通してるかテスターで調査
     → 右モーターのリレーの内、3接点中1箇所が同通不良
リレーを入れ替えてみる
     → 問題は変わらず

→ リレーのソケットが不良の可能性が高い

yasuohayashibara, 2025-06-02:
詳細なチェックありがとうございます．
電子回路側に問題があるのは予想外でした．
このような詳細なレポートは良いですね．
可視化されないとそもそも問題が発生したことを認識できないので，電子回路には問題が無いということを前提に考えてしまうので．

最後に擦れているような音がするというのは気になりました．
歴代のロボットを考えると，ギアが少しかけて交換することを繰り返してきましたので，そろそろかなという気がしています．
交換の必要がある場合は，おそらく納期が１～２ヶ月ぐらいの部品ですので，早めに動いたほうが良いと思います．

onlydcx, 2025-06-05:
修理をするにあたり、リレーのソケットの予備が見当たらなかった為4個購入しました。

onlydcx, 2025-06-16:
リレー基板を新品に交換後，モーターが正常に回転し屋外での自律走行も問題なくできました。