open-rdc/orne-box #130: 屋外で記録したrosbagから地図を生成するとノイズ?が生じる

YuseiShiozawa, 2025-04-23:
屋外でコントローラを用いて走行させて記録したrosbagから2d地図を作成すると、ノイズのようなものがロボットの軌跡をたどるように発生した。
作成にあたって使用している/scanはhokuyolidar(最大30m)から得ている。
また、 #95 の2d地図(屋外)でもノイズのようなものが発生していることを確認できる。

以下試したこと
- 屋内(2号館3階)で試す
  - ノイズのようなものは発生しない 
- rosbagを再生するスピードを落とす
  - 0.5倍, 0.01倍にしたが変化なし
- rosbag再生時のclockオプションの有無
  - 変化なし
- 地図作成ツールの変更
  - cartographer: 発生しない
  - gmapping: 発生する
  - slam_toolbox: 発生する

[image]

Ryusei-Baba, 2025-04-23:
ロボットのすぐ横に人がいる状態で（2D LiDARに人が映り込んでいる状態で）rosbagを取得してしまった可能性はありませんか？

https://github.com/open-rdc/orne-box/issues/95 については、R-Fans（360度のLiDAR）を使用して実験を行っているため、ロボットの後方に人が同行しており、その人が3DLiDARに映っている状態だったと考えられます。

rosbagに原因があるのか、それとも地図作成ツール側に問題があるのかを切り分けるために、別の屋外環境で取得したrosbagデータを使って試してみてください。

Ryusei-Baba, 2025-04-23:
Cartographerには、このような一時的なノイズを自動で除去するようなパラメータ設定があったようにも記憶していますが、詳細は定かではありません。

YuseiShiozawa, 2025-04-23:
> Cartographerには、このような一時的なノイズを自動で除去するようなパラメータ設定があったようにも記憶していますが、詳細は定かではありません。

こちらパラメータの数がかなり多く、そもそもの正確な地図生成を行うことに苦戦しています。

yasuohayashibara, 2025-04-24:
一部でも動画を載せてくれると何か分かるかもしれません．
GitHubに貼ると容量オーバーになるので，（卒業しても消えないアカウントの）YouTubeかGoogle Driveが良いです．
LiDARは直射日光に弱いので，そのせいかもしれません．

YuseiShiozawa, 2025-04-24:
> 一部でも動画を載せてくれると何か分かるかもしれません． GitHubに貼ると容量オーバーになるので，（卒業しても消えないアカウントの）YouTubeかGoogle Driveが良いです． LiDARは直射日光に弱いので，そのせいかもしれません．

[slam_toolboxで地図作成](https://youtu.be/etLtZoO-98I?si=OORnvs11xUywsRCb)
こちら津田沼キャンパス一周分の動画です。
日光に関しては考慮していませんでした。夜にrosbagを記録することも試そうと思います。

YuseiShiozawa, 2025-04-24:
footprintが間違っているのであれば、屋内でも同じ現象が起こると思います。
また、slam_toolboxには`footprint`含め、ロボットの大きさを定義するパラメータはありませんでした。

YuseiShiozawa, 2025-04-24:
/scan自体にはノイズのようなものは見られません。

[image]

YuseiShiozawa, 2025-04-24:
> ロボットのすぐ横に人がいる状態で（2D LiDARに人が映り込んでいる状態で）rosbagを取得してしまった可能性はありませんか？
> 
> [#95](https://github.com/open-rdc/orne-box/issues/95) については、R-Fans（360度のLiDAR）を使用して実験を行っているため、ロボットの後方に人が同行しており、その人が3DLiDARに映っている状態だったと考えられます。
> 
> rosbagに原因があるのか、それとも地図作成ツール側に問題があるのかを切り分けるために、別の屋外環境で取得したrosbagデータを使って試してみてください。

こちら2024つくばチャレンジのrosbagで作成したものですが、同じ現象が起こりました。
[image]

YuseiShiozawa, 2025-04-24:
/rfans/surestar_scan(point_cloudからlaserscanに変換したもの)で地図を作ると、かなりノイズらしきものを軽減することができた。しかし拡大すると、まだそれは点在していることが分かる。

[image]

yasuohayashibara, 2025-04-24:
なるほどです．
ただし，footprintは割と設定されていないことが多いパラメータなので，確認することのリストに加えておいてください．
こちらよく見ると経路上に【計測した】ノイズがあるのではなく，【計測されていない】空間があるという感じですね．
LiDARは近いものは計測しない（できない）のでそれが影響しているかもしれません．

yasuohayashibara, 2025-04-24:
あと，最近のGitHubの仕様をよく把握していませんが，動画を貼ると容量オーバーになるおそれがあるかもしれませんので，お気をつけください．
YouTubeなどにアップロードしてリンクを張ったほう良いです．

yasuohayashibara, 2025-04-27:
こちら，屋内と屋外で挙動が異なることを踏まえて考え直したのですが，おそらく以下が関連していると思います．
実は検討後に「そういえば」と思い，ChatGPTに画像を入れて質問したところ，ずばりの回答が返ってきました．
ROSに関しては，とりあえずChatGPTに聞くのが良いかなと改めて思いました．

-----

<html>
<body>
<h1 data-start="57" data-end="63" class="">📍現象</h1>
<ul data-start="64" data-end="170">
<li data-start="64" data-end="113" class="">
<p data-start="66" data-end="113" class=""><strong data-start="66" data-end="113">経路に沿って直線的に「未検出の領域（dark grayのunknown）」が残っている</strong></p>
</li>
<li data-start="114" data-end="138" class="">
<p data-start="116" data-end="138" class=""><strong data-start="116" data-end="127">屋外だけで発生</strong>（屋内では発生しない）</p>
</li>
<li data-start="139" data-end="170" class="">
<p data-start="141" data-end="170" class=""><strong data-start="141" data-end="164">スキャンの方向に広がる筋状の未検出領域</strong>になっている</p>
</li>
</ul>
<hr data-start="172" data-end="175" class="">
<h1 data-start="177" data-end="193" class="">📍これは何が起きているか？</h1>
<p data-start="195" data-end="257" class="">→ <strong data-start="197" data-end="246">LiDARのビームが無限遠(inf)になったが、地図上で"空間を埋める処理ができなかった"</strong><br data-start="246" data-end="249">
という現象です．</p>
<p data-start="259" data-end="277" class="">理由は次のいずれか，または複合です：</p>
<div class="group pointer-events-none relative flex justify-center *:pointer-events-auto"><span class="pointer-events-none absolute start-full top-4 z-10 hidden h-full w-fit ps-2 md:block" data-state="closed"><button class="hover:bg-token-main-surface-secondary text-token-text-secondary pointer-events-auto rounded-lg px-1 py-1 opacity-0 transition-opacity duration-200 group-focus-within:opacity-100 group-hover:opacity-100"><svg width="24" height="24" viewBox="0 0 24 24" fill="none" xmlns="http://www.w3.org/2000/svg" class="icon-md-heavy"><path fill-rule="evenodd" clip-rule="evenodd" d="M7 5C7 3.34315 8.34315 2 10 2H19C20.6569 2 22 3.34315 22 5V14C22 15.6569 20.6569 17 19 17H17V19C17 20.6569 15.6569 22 14 22H5C3.34315 22 2 20.6569 2 19V10C2 8.34315 3.34315 7 5 7H7V5ZM9 7H14C15.6569 7 17 8.34315 17 10V15H19C19.5523 15 20 14.5523 20 14V5C20 4.44772 19.5523 4 19 4H10C9.44772 4 9 4.44772 9 5V7ZM5 9C4.44772 9 4 9.44772 4 10V19C4 19.5523 4.44772 20 5 20H14C14.5523 20 15 19.5523 15 19V10C15 9.44772 14.5523 9 14 9H5Z" fill="currentColor"></path></svg></button></span><div class="tableContainer horzScrollShadows relative">
原因 | 説明 | 対策
-- | -- | --
inf_is_valid: falseになっている | infを無効と扱い、free spaceを広げなかった | inf_is_valid: trueにする
SLAM側の「最大ビーム長設定」が短い | 例：maxRangeやmaxUrangeが実際のセンサレンジより短い | maxRangeを広げる（例：20m→30m）
センサ自体のLiDAR最大距離が足りない | 屋外では対象物が遠すぎてそもそも反射してこない | LiDAR設定の最大レンジを調整、または高性能LiDAR使用検討

</div></div>
<hr data-start="583" data-end="586" class="">
<h1 data-start="588" data-end="604" class="">📍なぜ屋外で発生するのか？</h1>
<ul data-start="606" data-end="747">
<li data-start="606" data-end="647" class="">
<p data-start="608" data-end="647" class="">屋外では<strong data-start="612" data-end="623">建物や壁が遠い</strong>ため，LiDARビームが届かず，infが多くなる</p>
</li>
<li data-start="648" data-end="720" class="">
<p data-start="650" data-end="720" class="">infを「有効」と認識しない設定（<code data-start="667" data-end="688">inf_is_valid: false</code>）になっていると，<strong data-start="697" data-end="713">その方向がunknown</strong>になってしまう</p>
</li>
<li data-start="721" data-end="747" class="">
<p data-start="723" data-end="747" class="">屋内では近距離で壁に当たるので，問題が目立たない</p>
</li>
</ul>
<hr data-start="749" data-end="752" class="">
<h1 data-start="754" data-end="763" class="">📍対策まとめ</h1>
<h3 data-start="765" data-end="779" class="">【まずやるべきこと】</h3>
<p data-start="780" data-end="841" class="">✅ LiDARのドライバ設定・launchファイルで<br data-start="806" data-end="809">
　→ <strong data-start="812" data-end="841"><code data-start="814" data-end="834">inf_is_valid: true</code>に設定する</strong></p>
<h3 data-start="843" data-end="859" class="">【併せて確認したいこと】</h3>
<p data-start="860" data-end="873" class="">✅ SLAMのパラメータで</p>
<ul data-start="874" data-end="953">
<li data-start="874" data-end="906" class="">
<p data-start="876" data-end="906" class=""><strong data-start="876" data-end="888">maxRange</strong>（または<strong data-start="892" data-end="905">maxUrange</strong>）</p>
</li>
<li data-start="907" data-end="953" class="">
<p data-start="909" data-end="953" class=""><strong data-start="909" data-end="936">missing_data_ray_length</strong>（cartographerの場合）</p>
</li>
</ul>
<p data-start="955" data-end="968" class="">が小さすぎないかチェック．</p>
<p data-start="970" data-end="1003" class="">→ センサー実測値に合わせて，例えば <strong data-start="989" data-end="996">30m</strong> とかに設定．</p>
</body>
</html>

YuseiShiozawa, 2025-04-28:
ありがとうございます。
しかし、[urg_node](https://github.com/ros-drivers/urg_node/blob/melodic-devel/launch/urg_lidar.launch)には、`inf_is_valid`を有効にするか否かのパラメータがないと思います。
また、LiDARのビームが跳ね返ってこないときはinfにはならず、[本体の仕様](https://www.monotaro.com/g/02608656/?srsltid=AfmBOorSQ2vnEXxXt0hBnHQsIgv5lEvdQAPxE4hiuvqfZoDYeoCWx7HG)通り、60(m)という値が入ることを確認しました。この60(m)という値はパラメータで変更できず、固定値になっています。 

そして、slamでの最大検出距離のパラメータは、特に小さい値に設定せず、lidarの検出距離通り30(m)としています。 

私もこの件についてchatGPTに改めて聞くと、先生と同様の内容が返って来ました。しかし、しつこく問い合わせたら添付した画像の結果となりました。

- `rostopic echo /scan` の一部
```
header: 
  seq: 8730
  stamp: 
    secs: 1744952465
    nsecs: 501735055
  frame_id: "hokuyo_link"
angle_min: -1.5707963705062866
angle_max: 1.5707963705062866
angle_increment: 0.004363323096185923
time_increment: 1.736111516947858e-05
scan_time: 0.02500000037252903
range_min: 0.023000000044703484
range_max: 60.0  <-- これ

```

[image]

YuseiShiozawa, 2025-04-28:
pointcloud_to_laserscanを以下の設定にしてslam_toolboxで地図を作成した。
結果は下図のようにノイズらしきものは消え、おまけに地図の正確さも向上した。200m先まで見ているので正確さが上がることは概ね予想通りだが、ノイズが消えた理由はあまりよく分からない。

```
<launch>
  <node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan" output="screen">
    <remap from="cloud_in" to="rfans/surestar_points"/>
    <remap from="scan" to="tmp_scan"/>

    <param name="min_height" value="0.0"/>
    <param name="max_height" value="4.0"/>

    <param name="range_min" value="0.0"/>
    <param name="range_max" value="200.0"/>

    <param name="angle_min" value="-1.57"/>
    <param name="angle_max" value="1.57"/>
    <param name="angle_increment" value="0.0035"/>
  </node>
</launch>

```
[image]

closed 2025-05-22 by Ryusei-Baba