open-rdc/orne-box #132: SSD変更によるCPU負荷の軽減の検討

Shuma-s58, 2025-05-20:
現状のbox3のSSDの書き込み速度は以下のようになっている.

[image]

1GBを1回書き込み　1200MB/s   (0.871秒)
1GBを30回書き込み　1400MB/s  (22.94秒)

このSSDをGen4.0に変更することにより, 書き込み速度が最大3400 MB/s (2.8倍) になる可能性がある.
(jetson orin nano側の規格はGen3.0であるが, Gen4.0には下位互換性があり, Gen3.0の性能限界近い値を出せる可能性がある)

書き込み速度が遅いとCPU側に処理がたまり, CPU使用率が上がる原因となるため, 
SSDの変更により, CPU負荷を軽減できるのではないかと考える.

この変更に有効性があるか引き続き調査を行っていく.

Shuma-s58, 2025-05-20:
何か調べるべきことやアドバイスなどありましたら, お願いします.

yasuohayashibara, 2025-05-20:
できれば現状のシステム構成を教えて下さい．

あと，LLMに質問した結果を貼ります．

高速SSDへの変更は有効ですが、以下の組み合わせが最適：
- 
- NVMe SSDの採用
- rosbag2のfastwriteプロファイル適用
- メモリキャッシュサイズの調整（--max-cache-size 1GB以上）
- 不要なデータ圧縮の無効化

これにより、従来比で[3](https://github.com/ros-tooling/rosbag2_storage_mcap)-5倍の書き込み速度改善が期待できます。ただし、極端に高いデータレート（例：1GB/s超）を要求する場合、ROSのアーキテクチャ自体の見直しが必要になる可能性があります。

yasuohayashibara, 2025-05-20:
以下の命令でSSD読み書き次のCPUの負荷を確認することができるとのことです．
（間接的にという表現がされていますが．．．）

```
iostat -x 1
```