open-rdc/orne-box #133: box3　Rvizにマップが表示されない

yuukitsubouchi, 2025-05-20:
津田沼チャレンジ以降、ブランチhumble/tsudanuma2025でのソースコードを一切変えてないのに、BOX3を動かした時に、Rvizにマップが表示されないので困ってます

yasuohayashibara, 2025-05-20:
分かっている人を呼んで助けてもらうのが良いとは思いますが，まずは確認したほうが良いのは以下ぐらいでしょうか．

1) まずは`git diff`で何か変更がないかを確認してください．
2) rvizで地図を表示する設定になっているかを確認してください．
3) 再ビルドしてください．

yuukitsubouchi, 2025-05-22:
1 変更は確認しましたが、変更したのを戻しても表示されない
2 rvizで地図を表示する設定になっている
3 再ビルドしたが表示されなかった

yasuohayashibara, 2025-05-22:
emcl2でTFに問題があると出ていますね．
必要なTFが出力されていないか，出力する前にemcl2が起動したという可能性があります．
タイミングの問題の可能性があるので，一つ一つノードを起動して，動作を確認していくと解決しそうな気がします．

ちなみに，こちらのソフトウェア構成を把握していませんが，林原が作成したシステムでは，amclとemclをオプションで切り替えられるようにしていましたので，amclでも良ければそちらを利用するという方法があります．

yuukitsubouchi, 2025-05-23:
Rfansのlaunchファイルを単体で起動したところ surestar_pointsのtopicの中身が空でした。(/scanと/surestarも空)
ハードウェア(3DRidar)に問題がある可能性があります

yasuohayashibara, 2025-05-23:
誰かわかる方に見てもらったほうが良いですね．
電子回路側をチェックする場合は，電子回路がよく分かる方がいたほうが良いです．

yuukitsubouchi, 2025-05-23:
ping192.168.0.3をしたところ、destination host unreachable が出てきました

Ryusei-Baba, 2025-05-23:
ハードウェアの問題というよりは、ネットワークのプロファイルに問題があると思います。

Ubuntuの設定画面から有線LANの設定を再度行ってみてください。

Nomura1024, 2025-05-28:
点群がパブリッシュされないため，内部の配線を疑って調査を行った．
r-fans付属のケーブルにある電源スイッチ付近で断線しているような様子が見られたが，分割をしてみると接続されていそうだった．しかし，ケーブルの固定方法がとても不安定なため今後断線はありえそうである．スイッチの中を確認し元に戻した後，再度動作確認をしたところ正常に点群がパブリッシュされるようになった．また，馬場君がネットワークのプロファイルの確認をしてもらったが問題がなそうであった．
その後，屋外で走行させたが問題なかったとのことなのでしばらく様子見することにする．
スイッチが問題と断定できないが，今後同じ問題が起こるようであればスイッチはほぼ使っていないためスイッチをコネクタに変更を考えている．
[image]
[image]

masato4988, 2025-06-20:
問題が再発しました。
このiaaueの内容は試しましたが、トピックがパブリッシュされていないようです

masato4988, 2025-06-20:
ping 192.168.0.3
をしたところ通信しているように見えたのですが、
実際はrfansはIPが192.168.0.3ではないもので接続されていたようです。
ネットワークの設定をし直したところ192.168.0.3でrfansと接続することができました。

closed 2025-07-11 by Ryusei-Baba